tzaeschke/ode4j #29: Trimesh to Trimesh Collisions

jfelrod1960, 2015-07-13:
What can I do (in general) to improve these types of collisions?  I have a vehicle and a pilot that will sit inside the vehicle.  The cockpit in the vehicle is open so the pilot is visible and I plan to animate the pilot while inside the vehicle.  But it should have a "ragdoll" behavior when colliding with other objects.  So I can't use a box geom for the vehicle or it will be difficult to place the pilot inside the vehicle.

Thanks!
Jeff

tzaeschke, 2015-07-14:
Sorry, I'm a bit confused, the title mentions trimesh-trimesh collision, but the question is about ragdoll?

I don;t have experience with modelling cars, but here are some thoughts. First, I think you could disable collision between ragdoll and Box, so the ragdoll could be put inside (even anchored to?) the box.
But then the ragdoll could still svere a lot to the sides, even outside the box. For more realistic physics, you could probably build a 'hollow' box out of five boxes, one for the floor and 4 for the sides. 
But there may be smarter solutions.

You could also ask the question on the classic ODE forum. ODE doesn't have ragdolls, but they have a lot of experience and may be able to solve your problem.

Btw, please have a look at issue #30. I believe such questions should be made on a forum, so I think we should finally create one.

jfelrod1960, 2015-07-14:
Tilmann I'm sorry for the confusion.  I want to use a trimesh geom for my vehicle.  This will allow me to seat the pilot (which is a kinematic object) inside the open areas of the geom.  But colliding with the other vehicles that are represented by a trimesh geom make performance degradation severe.  I hope that clarifies.  Now with that said I may try the hollow box suggestion and not use a trimesh.

piotr-piastucki, 2015-07-14:
I am surprised as I have not really noticed such a severe performance degradation. See the following video for instance - https://vimeo.com/133474815. Suzie model I used consists of  564 vertices and 968 faces which is quite a lot for a collision mesh IMHO. Of course the performance drops at the end but there are 15 meshes colliding with each other. How complex is your car collision mesh then?

jfelrod1960, 2015-07-15:
@piotr-piastucki Sorry for the late response.  I didn't see the email notification until last night.  The model I use has 11022 vertices and 19679 faces.  My game will have about 20 vehicles and the steering behavior framework should keep them from colliding as much with each other.  I may have to get low poly versions of the vehicles.  With box geoms they work well so I plan to follow Tilmann's advice.  I would appreciate any suggestion you have.  Thanks for replying. :)

chriscamacho, 2015-07-15:
@jfelrod1960 you really need to have collision proxies - you shouldn't really be just throwing a complex triangle soup to a collision library, fine for detailed visual representations but crippling for collisions! 

Make a limited number of primitive geometries (box etc) for the car proxy, if you want the driver to rag doll when you do a rapid break, then only introduce the rag doll model when you determine a collision is violent enough to have ejected the driver from the car...

I have in the past used cylinders for tires for a 4x4 vehicle and even 64 triangles for tread mesh (32 rects made from tri's) any higher detail will lead to slow downs in any collision library.

This isn't really an ide4j bug - you're asking for basic advice on game programming...

jfelrod1960, 2015-07-15:
@chriscamacho Thanks for replying.  I am new to game development so I appreciate the advice.  I did enter this as an issue, but only to get some advice on optimizing trimesh to trimesh collisions.  You are totally correct, this is not an ode4j bug.  For now I'm just using a box geom for the vehicle.  It would be cool to have a forum.  Just my two cents. :)

chriscamacho, 2015-07-15:
don't forget that a body can have mutiple geoms (shapes like boxes or spheres) and I'm sure that when Ode4J does get a forum there will be plenty of people happy to give advice...

closed 2017-01-18 by jfelrod1960