tzaeschke/ode4j #30: Forum? Feedback requested

tzaeschke, 2015-07-14:
I believe we need a forum or something similar.
There are many options:

- Google Forums
- StackOverflow
- Anything else?

What is you opinion on these options? Any other proposals?
Google forums would be a classical forums to discuss topics in length. However Stackoverflow would make it easier to cross-link related topics (they also list related topics in the sidebar).
What do you think?

chriscamacho, 2015-07-14:
something that doesn't require a social media account would be nice...

tzaeschke, 2015-07-14:
@chriscamacho Well, any forum is a social media, isn't it? :-)
Google comes with Google Hangouts I guess. 
StackOverflow can be used without Facebook or similar, but registration is a bit hidden. So I take this as a vote for StackOverflow?
Any other suggestions?

kephale, 2015-07-14:
I've started using gitter, which does require a github account, but that is it. It integrates with a github repository and makes a chatroom, recording the chat history (i haven't checked to see how long the history is saved, but i expect indefinitely). Folks are pretty into Slack now, which I've been happy with (you can get notifications on repository updates within chatrooms). Gitter seems to have the least overhead in terms of requiring extra accounts though.

jfelrod1960, 2015-07-14:
Personally I vote for google groups.  I would like that better than using the regular ODE group because a lot of questions may also depend on Java performance as well as optimizing for collisions and things like that.

tzaeschke, 2015-07-16:
gitter looks nice, but I think it's more suited for conversation-like exchange.
I think a forum has the advantage that topics are separated by threads and thus easier searchable.
So I count the following so far:

- Google forums: 2 votes (jfelrod & me)
- gitter: 1 vote (kephale)
- StackOverflow: 0 votes (?)

I think none of these really counts as social media accounts (Twitter, Facebook, ...), would you agree?

Unless the voting changes a lot soon (or anyone has severe objections) I will set up google forums, probably this weekend.

chriscamacho, 2015-07-16:
A vps can be had very cheaply nowadays and you can throw up your own solution easily with minimal maintenance

why assume everyone will have or even want g+ accounts from the evil empire, or similar vast social media company, you didn't count my vote AGAINST Google forums !!

piotr-piastucki, 2015-07-16:
Taking into account that the official ode-users forum is also hosted at Google it makes a lot of sense to go for Google Groups, this way you will have a single place for all ODE-related discussions.

chriscamacho, 2015-07-16:
which is why I and others don't bother with it...

kephale, 2015-07-16:
Seconding Google Groups.

tzaeschke, 2015-07-16:
@chriscamacho 
Sorry, it wasn't clear to me that you count Google as social media, for me it's mainly email and forum. I don't think many people really 'want' a google account, but many have one already and it may be the easiest option overall. Personally, I use my google account almost exclusively for google groups...

So the votes so far are:

- Google forums: +3 votes (jfelrod & me, kephale)  - 1 (chriscamacho)
- gitter: (1 vote?) (kephale)
- StackOverflow: 0 votes
- VPS: 1 vote (but we need a volunteer to host it)

I don't mind the VPS option if someone would take care of it (ideally for years to come)?

chriscamacho, 2015-07-16:
I'd be more than happy to maintain a VPS for years to come - or find a replacement if needed - if it were funded...

tzaeschke, 2015-07-16:
Funding anyone? Concerns?

jfelrod1960, 2015-07-16:
A VPS would be great if we had a large number of ode4j users, but from what I can tell, there is only a few of us.  Having some kind of forum may attract more users to ode4j.  We may eventually get to a time when we could get enough people to donate through patreon or whatever and fund a VPS.  But we have to start somewhere.  I really enjoyed the help I received from my question and a forum of some kind would be a great tool for others seeking help.

kephale, 2015-07-16:
I still vote Google Groups, and would probably be even more likely to vote for Groups with a larger user base. 

That being said, a point in favor of a VPS, is less about using it for forums, but instead to use it as a continuous integration server for providing a continuous ode4j release (and maybe acting as a maven repository). 

I would be happy to help with applying for funding. It is probably the type of thing that could be well served by small grants, or even crowd funding.

tzaeschke, 2015-07-20:
I created a forum on Google Groups:
https://groups.google.com/forum/?hl=en#!forum/ode4j

@chriscamacho Sorry, Google Groups seems to be the preferred choice and easiest to implement for me.
If we get around to set up a VPS (maybe with build service), or any other good option, I'm happy to move the forum somewhere else.

chriscamacho, 2015-07-20:
alas I just don't trust them (google) any more...

closed 2015-07-27 by tzaeschke