tzaeschke/ode4j #36: Garbage collection - step()

tzaeschke, 2015-10-06:
As mentioned in issue #35, ode4j appears to allocate significant amounts of memory in some situations/systems ("4-5 MB/s"). 

The problem may be in the stepper which allocates large arrays for the solver during each step.

A solution could involve array pooling.

tzaeschke, 2015-10-29:
I implemented an experimental hack which can be enabled with:
`DxQuickStep.REUSE_OBJECTS = true;`

This reuses a number of objects, including the large collision used in the stepper.

Could someone have a look whether it improves the situation on Android? We could apply the same approach to other places in the code.

piotr-piastucki, 2015-12-08:
It looks like some improvement has been recently made in ODE too - see https://bitbucket.org/odedevs/ode/commits/506a3b3b93c2ef0da3e5030bf12e408aef59c2cd

tzaeschke, 2015-12-08:
Yes, they seem to avoid allocating `Ad`, even if that is much smaller than other matrices (`iMJ` is 12 times bigger). 
What I did in this patch is really just a hack and should definitely be extended to the matrix solver as well.
I also looked a few days ago at porting ODE changes to ode4j. But quickstep has changed considerably, so I suppose I need a week off. Maybe over the upcoming holidays.

Has anyone seen a measurably reduced GC load from my initial patch?