tzaeschke/ode4j #63: Trouble compiling to Java 7/8

jfelrod1960, 2018-03-26:
From the comments that were made, I understood that we can compile ode4j to Java 7.  But I'm having trouble compiling to Java 7.  Can someone explain how with the new changes in place?

tzaeschke, 2018-03-26:
The actual Java code should compile just fine with Java 7. I suspect Maven complains about the Java 9 requirement in the pom?

Can you tell us why you need Java 7 source code? Are you compiling for Android?

jfelrod1960, 2018-03-26:
I'm using ode4j in Android, but I'm having trouble running the demos and with maven I get the following:

[INFO] Scanning for projects...
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO]
[INFO] ODE for Java
[INFO] core
[INFO] core-cpp
[INFO] demo
[INFO] demo-cpp
[INFO]
[INFO] ------------------------------------------------------------------------
[INFO] Building ODE for Java 0.4.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO]
[INFO] --- maven-clean-plugin:3.0.0:clean (default-clean) @ parent ---
[INFO] Deleting C:\Users\jfe.CAMPUS\Documents\GitHub\ode4j\target
[INFO]
[INFO] --- maven-enforcer-plugin:1.0:enforce (enforce-maven) @ parent ---
[INFO]
[INFO] --- maven-enforcer-plugin:1.0:enforce (enforce-rules) @ parent ---
[WARNING] Rule 0: org.apache.maven.plugins.enforcer.RequireJavaVersion failed with message:
Detected JDK Version: 1.8.0-121 is not in the allowed range [9,).
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO]
[INFO] ODE for Java ....................................... FAILURE [  2.706 s]
[INFO] core ............................................... SKIPPED
[INFO] core-cpp ........................................... SKIPPED
[INFO] demo ............................................... SKIPPED
[INFO] demo-cpp ........................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 2.920 s
[INFO] Finished at: 2018-03-26T14:30:50-04:00
[INFO] Final Memory: 9M/245M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-enforcer-plugin:1.0:enforce (enforce-rules) on project parent: Some Enforcer rules have failed. Look above for specific messages explaining why the rule failed. -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

Thanks

tzaeschke, 2018-03-26:
It seems you are using Java 8, would it work for you to use Java 9? 
If you are already on Java 8, Java 9 should be okay as well?

ppiastucki, 2018-03-26:
I can confirm it compiles fine with Java 9. Taking into account Oracle's plan to release new Java version every 6 months switching to Java 9 in the near future is probably inevitable anyway.

jfelrod1960, 2018-03-27:
I'm using Java 8 which is fine with Android 7.1 and above.  Switching to 9 will not work for Android.  At least not now.  Maybe future versions if Google chooses.

io7m, 2018-03-27:
> Switching to 9 will not work for Android. At least not now.

Are you sure about this? You're right that Android can't run Java 9 bytecode, but you should be able to use JDK 9 to compile the code (producing Java 7-8 bytecode).

jfelrod1960, 2018-03-27:
I do not want to switch to Java 9 JDK on my machine right now.  If I have to stay with Java 8 for other reasons.  If the decision is to move ode4j to Java 9 then I will respect that decision.  I will have to seek other alternatives.

io7m, 2018-03-27:
Don't worry, there is another way.

I can submit some changes to the build system such that if you build the code on a JDK <= 8, it'll build but you won't get modularization. If you build it on a JDK >= 9, you'll get modularization. It is doable, but it's more complex (and hence why I didn't do it that way the first time).

jfelrod1960, 2018-03-27:
I have to be honest with you.  I have some concerns about the performance using ode4j in Android.  If performance in an issue then my only other choice is to make my game a desktop game.  You are very kind to work with me on this, but my game may end up on the desktop.  At that point, it would not matter which version of Java to use.

tzaeschke, 2018-03-27:
Travis.ci was already working, I now added a badge to the README.md.

I think we definitely should _not_ move forward if we push away users, especially since the Java 9 compatibility is at the moment more something of a convenience feature.

However, if there is a way to build this with Java 8 or even Java 7, that would be very cool.

Another question would be if there is anything we could help with moving to Java 9, is it something related to ode4j and/or Android? I have Java 6, 7, 8 and 9 installed in parallel on my machine, no problem (mostly using Eclipse, but also IntelliJ and NetBeans installed).

io7m, 2018-03-27:
> I think we definitely should not move forward if we push away users, especially since the Java 9 compatibility is at the moment more something of a convenience feature.

I agree. Modularization is good (consider `jlink` for deployment and so on), but excluding as few users as possible takes priority (for me at least).

jfelrod1960, 2018-03-27:
I want java to default to 8 on my machine.  If I can install 9 on my machine and get maven to use 9 instead of 8 then that would suffice.  I'm not a maven expert since I use gradle for my builds.  Please advise?

io7m, 2018-03-27:
> I installed Java 9 on my machine. I'm still having trouble. In intellij it thinks the source level is set up to 5. Errors using the diamond operator. I may use Piotr version for now.

Two things:

1. I think the simplest way to get Maven to use JDK 9 for builds but leave you with Java 8 as the default JDK is to set `JAVA_HOME` and to run the specific `java` binary you want. It's not different from other Java programs in that regard:

```
$ java -version
openjdk version "1.8.0_162"
OpenJDK Runtime Environment (build 1.8.0_162-b12)
OpenJDK 64-Bit Server VM (build 25.162-b12, mixed mode)

$ which java
/usr/bin/java

$ /usr/lib/jvm/java-9-openjdk/bin/java -version
openjdk version "9.0.4"
OpenJDK Runtime Environment (build 9.0.4+11)
OpenJDK 64-Bit Server VM (build 9.0.4+11, mixed mode)

$ JAVA_HOME=/usr/lib/jvm/java-9-openjdk mvn clean package
[INFO] Scanning for projects...
```

The problem you're having with Intellij is a known issue that I think has an easy fix (I'm about to submit it). It expects to see a `<source>` entry in the POM file to determine what version of Java source code to allow, but our current POM only has `<release>`. The `<source>` entry is actually deprecated now, but IDEA hasn't been updated to recognize `<release`> yet.
```

jfelrod1960, 2018-03-28:
Found in intellij what I did wrong.  But installing Java 9 on my machine broke another application.  I removed the jre, but kept the jdk.  Will update you on using maven when I get a chance.

jfelrod1960, 2018-03-28:
I'm getting the following error when trying to generate javadocs

[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 8.718 s
[INFO] Finished at: 2018-03-28T16:27:17-04:00
[INFO] Final Memory: 17M/58M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-javadoc-plugin:2.10.3:javadoc (default-cli) on project core: An error has occurred in JavaDocs report generation:
[ERROR] Exit code: 1 - C:\Users\jfe.CAMPUS\Documents\GitHub\ode4j\core\src\main\java\module-info.java:28: error: module not found: org.slf4j
[ERROR] requires org.slf4j;
[ERROR] ^
[ERROR]
[ERROR] Command line was: "C:\Program Files\Java\jdk-9.0.4\bin\javadoc.exe" @options @packages @argfile
[ERROR]
[ERROR] Refer to the generated Javadoc files in 'C:\Users\jfe.CAMPUS\Documents\GitHub\ode4j\core\target\site\apidocs' dir.
[ERROR] -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.apache.maven.plugins:maven-javadoc-plugin:2.10.3:javadoc (default-cli) on project core: An error has occurred in JavaDocs report generation:
Exit code: 1 - C:\Users\jfe.CAMPUS\Documents\GitHub\ode4j\core\src\main\java\module-info.java:28: error: module not found: org.slf4j
    requires org.slf4j;
                ^

Command line was: "C:\Program Files\Java\jdk-9.0.4\bin\javadoc.exe" @options @packages @argfile

Refer to the generated Javadoc files in 'C:\Users\jfe.CAMPUS\Documents\GitHub\ode4j\core\target\site\apidocs' dir.

        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:212)
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:153)
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:145)
        at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject(LifecycleModuleBuilder.java:116)
        at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject(LifecycleModuleBuilder.java:80)
        at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build(SingleThreadedBuilder.java:51)
        at org.apache.maven.lifecycle.internal.LifecycleStarter.execute(LifecycleStarter.java:128)
        at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:307)
        at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:193)
        at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:106)
        at org.apache.maven.cli.MavenCli.execute(MavenCli.java:863)
        at org.apache.maven.cli.MavenCli.doMain(MavenCli.java:288)
        at org.apache.maven.cli.MavenCli.main(MavenCli.java:199)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:564)
        at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Launcher.java:289)
        at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.java:229)
        at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(Launcher.java:415)
        at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:356)
Caused by: org.apache.maven.plugin.MojoExecutionException: An error has occurred in JavaDocs report generation:
Exit code: 1 - C:\Users\jfe.CAMPUS\Documents\GitHub\ode4j\core\src\main\java\module-info.java:28: error: module not found: org.slf4j
    requires org.slf4j;
                ^

Command line was: "C:\Program Files\Java\jdk-9.0.4\bin\javadoc.exe" @options @packages @argfile

Refer to the generated Javadoc files in 'C:\Users\jfe.CAMPUS\Documents\GitHub\ode4j\core\target\site\apidocs' dir.

        at org.apache.maven.plugin.javadoc.AbstractJavadocMojo.failOnError(AbstractJavadocMojo.java:6018)
        at org.apache.maven.plugin.javadoc.JavadocReport.execute(JavadocReport.java:322)
        at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo(DefaultBuildPluginManager.java:134)
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:207)
        ... 20 more
Caused by: org.apache.maven.reporting.MavenReportException:
Exit code: 1 - C:\Users\jfe.CAMPUS\Documents\GitHub\ode4j\core\src\main\java\module-info.java:28: error: module not found: org.slf4j
    requires org.slf4j;
                ^

Command line was: "C:\Program Files\Java\jdk-9.0.4\bin\javadoc.exe" @options @packages @argfile

Refer to the generated Javadoc files in 'C:\Users\jfe.CAMPUS\Documents\GitHub\ode4j\core\target\site\apidocs' dir.

        at org.apache.maven.plugin.javadoc.AbstractJavadocMojo.executeJavadocCommandLine(AbstractJavadocMojo.java:5163)
        at org.apache.maven.plugin.javadoc.AbstractJavadocMojo.executeReport(AbstractJavadocMojo.java:2075)
        at org.apache.maven.plugin.javadoc.JavadocReport.generate(JavadocReport.java:130)
        at org.apache.maven.plugin.javadoc.JavadocReport.execute(JavadocReport.java:318)
        ... 22 more
[ERROR]
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

jfelrod1960, 2018-03-28:
The Java 9 install really broke my build.  I'm using an entity-component system that uses annotation processing and now that won't compile my entity/component code.  This is going to require a lot of rework with things that was first working.  I should close this issue and not hold everyone else up.  Maybe I can investigate things further in the future.  I still have 0.3.1-SNAPSHOT for now, but that will lock me for now.  If the other developer is willing to make his entity/component work with Java 9 and if ode4j works with Android, then I can continue using this library.  Thanks for helping me.

io7m, 2018-03-28:
I think this should be undone if it's breaking people's builds. I haven't seen this in any of the projects I've assisted to date, but Android is always a killer.

There is another approach we can take to producing builds, I just haven't had time to submit patches yet.

io7m, 2018-03-28:
The JavaDoc issue is a bug in versions of the plugin prior to `3.0.0`. I'll upgrade the plugin in the `pom.xml` to compensate.

tzaeschke, 2018-03-28:
@io7m If you don't mind, I think I will roll back the pull-request and put it on a separate branch (not tonight, but soon). It's a useful change I hope we can merge it once use of Java 9 (also regarding Android) is a bit more widespread. Maybe we also get more feedback from other developers. What do you think?

io7m, 2018-03-28:
Yep, that sounds fine to me.

The other approach I have in mind uses Maven profiles and therefore won't bear much resemblance to the pure approach I took first of all. That'll work on Java 8 and below for certain, it's just a little ugly to look at. :sweat_smile:

jfelrod1960, 2018-03-29:
Can we keep the master branch as it is now and create a new 0.3.1 branch that conforms to Java versions below version 9?  That way I can submit the new changes from the master branch that will work in this new branch under versions below 9?  Seems like this is the better solution.  Don't you think?

tzaeschke, 2018-03-30:
I created a branch 'java9' with io7m's changes. I did a minimal  reversion of pullrequest #61, so that we keep almost all of the changes but make it build with Java 1.7 again. Even the .travis file is still on Java 9 :-).
@jfelrod1960 Could you try whether the new version works for you?

@io7m Even though we reverted parts of it, thanks a lot for your efforts, much appreciated! 

Maybe we can even get to the parallel Java 7/9 .pom file with Maven profiles at some point, lets see.

jfelrod1960, 2018-03-30:
@tzaeschke Yes it works.
@tzaeschke and @io7m I appreciate your kindness.  I feel like I'm holding up the progress of ode4j.  Anyway what you two did was very thoughtful and considerate.  Thank you.

tzaeschke, 2018-03-31:
@jfelrod1960 No worries, also, I suppose you may not be the Android developer (or otherwise bound to Java < 9) out there :-)

closed 2018-04-02 by jfelrod1960