tzaeschke/ode4j #69: Replace slf4j dependency

zeganstyl, 2020-07-30:
Can you remove slf4j dependency from core?

As I see it used only in `ErrorHdl` class.
Can you replace it with some interface, that can be implemented externaly, like as plugin?
For example you can create some global object like `Log`, which has static variable `imp` for storing logging interface implementation. This object may be used internaly in odej4. And users can implement any loggers that they like. For default implementation in core, can be simple `System.out.println`

slf4j internally using reflection, so I can't use ode4j with [teavm](https://github.com/konsoletyper/teavm).
Is odej4 using reflection?

Also I think the fewer dependencies the better.

zeganstyl, 2020-08-01:
TeaVM may use slf4j in compilation tools. It compiles JVM byte code to javascript.
To compile, I'm using TeaVM CLI tool.
As mentioned in [TeaVM docs](http://teavm.org/docs/intro/overview.html):
```
There are Java APIs that are impossible to implement without generating inefficient JavaScript.
Some of these APIs are: reflection, resources, class loaders, and JNI.
TeaVM restricts usage of these APIs.
```
These restrictions are for classes that must be compiled to javascript.

I have checked more details of compilation errors. There is also several errors with java.util.concurrent:
<details>
  <summary>Full details</summary>
  
  ```
INFO: Running TeaVM
Analyzing classes... complete for 8.918 seconds
Compiling... complete for 6.216 seconds
INFO: Output file built with errors
INFO: Debug information successfully written
INFO: Source maps successfully written
ERROR: Class java.util.concurrent.CountDownLatch was not found
    at org.ode4j.ode.threading.task.Task.<init>(Task.java:44)
    at org.ode4j.ode.threading.task.TaskGroup.<init>(TaskGroup.java:33)
    at org.ode4j.ode.threading.task.TaskGroup.<init>(TaskGroup.java:29)
    at org.ode4j.ode.threading.task.AbstractTaskExecutor.group(AbstractTaskExecutor.java:30)
    at org.ode4j.ode.internal.DxWorld.dxProcessIslands(DxWorld.java:705)
    at org.ode4j.ode.internal.DxWorld.dWorldQuickStep(DxWorld.java:360)
    at org.ode4j.ode.internal.DxWorld.quickStep(DxWorld.java:955)
    at org.ksdfv.thelema.phys.ode4j.OdePhysicsWorld.step(OdePhysicsWorld.kt:183)
    at org.ksdfv.thelema.test.phys.BoxOnBoxTest$testMain$1.invoke(BoxOnBoxTest.kt:95)
    at org.ksdfv.thelema.test.phys.BoxOnBoxTest$testMain$1.invoke(BoxOnBoxTest.kt:34)
    at org.ksdfv.thelema.gl.GL.doSingleCalls(GL.kt:273)
Class java.util.concurrent.CountDownLatch was not found

    at org.ksdfv.thelema.teavm.TeaVMApp$1.invoke(TeaVMApp.kt:113)
    at org.ksdfv.thelema.teavm.TeaVMApp$1.invoke(TeaVMApp.kt:38)
    at org.ksdfv.thelema.teavm.TeaVMApp$sam$org_teavm_jso_browser_AnimationFrameCallback$0.onAnimationFrame
    at org.ksdfv.thelema.teavm.TeaVMApp$sam$org_teavm_jso_browser_AnimationFrameCallback$0.onAnimationFrame$exported$0
ERROR: Class java.util.concurrent.CountDownLatch was not found
    at org.ode4j.ode.threading.task.Task.awaitCompletion(Task.java:50)
    at org.ode4j.ode.internal.DxWorld.dxProcessIslands(DxWorld.java:722)
    at org.ode4j.ode.internal.DxWorld.dWorldQuickStep(DxWorld.java:360)
    at org.ode4j.ode.internal.DxWorld.quickStep(DxWorld.java:955)
    at org.ksdfv.thelema.phys.ode4j.OdePhysicsWorld.step(OdePhysicsWorld.kt:183)
    at org.ksdfv.thelema.test.phys.BoxOnBoxTest$testMain$1.invoke(BoxOnBoxTest.kt:95)
    at org.ksdfv.thelema.test.phys.BoxOnBoxTest$testMain$1.invoke(BoxOnBoxTest.kt:34)
    at org.ksdfv.thelema.gl.GL.doSingleCalls(GL.kt:273)
    at org.ksdfv.thelema.teavm.TeaVMApp$1.invoke(TeaVMApp.kt:113)
    at org.ksdfv.thelema.teavm.TeaVMApp$1.invoke(TeaVMApp.kt:38)
    at org.ksdfv.thelema.teavm.TeaVMApp$sam$org_teavm_jso_browser_AnimationFrameCallback$0.onAnimationFrame
    at org.ksdfv.thelema.teavm.TeaVMApp$sam$org_teavm_jso_browser_AnimationFrameCallback$0.onAnimationFrame$exported$0
ERROR: Class java.util.concurrent.atomic.AtomicReference was not found
    at org.ode4j.ode.internal.processmem.DxWorldProcessContext.SetStepperArenasList(DxWorldProcessContext.java:65)
Class java.util.concurrent.atomic.AtomicReference was not found

    at org.ode4j.ode.internal.processmem.DxWorldProcessContext.ReallocateStepperMemArenas(DxWorldProcessContext.java:292)
    at org.ode4j.ode.internal.processmem.DxWorldProcessContext.dxReallocateWorldProcessContext(DxWorldProcessContext.java:400)
    at org.ode4j.ode.internal.DxWorld.dWorldQuickStep(DxWorld.java:356)
    at org.ode4j.ode.internal.DxWorld.quickStep(DxWorld.java:955)
    at org.ksdfv.thelema.phys.ode4j.OdePhysicsWorld.step(OdePhysicsWorld.kt:183)
    at org.ksdfv.thelema.test.phys.BoxOnBoxTest$testMain$1.invoke(BoxOnBoxTest.kt:95)
    at org.ksdfv.thelema.test.phys.BoxOnBoxTest$testMain$1.invoke(BoxOnBoxTest.kt:34)
    at org.ksdfv.thelema.gl.GL.doSingleCalls(GL.kt:273)
    at org.ksdfv.thelema.teavm.TeaVMApp$1.invoke(TeaVMApp.kt:113)
    at org.ksdfv.thelema.teavm.TeaVMApp$1.invoke(TeaVMApp.kt:38)
    at org.ksdfv.thelema.teavm.TeaVMApp$sam$org_teavm_jso_browser_AnimationFrameCallback$0.onAnimationFrame
    at org.ksdfv.thelema.teavm.TeaVMApp$sam$org_teavm_jso_browser_AnimationFrameCallback$0.onAnimationFrame$exported$0
ERROR: Class java.util.concurrent.atomic.AtomicReference was not found
    at org.ode4j.ode.internal.processmem.DxWorldProcessContext.GetStepperArenasList(DxWorldProcessContext.java:66)
[... 273 lines cut ...]

Method java.io.PrintStream.println(F)V was not found
  ```
</details>

<details>
  <summary>Briefly</summary>
  
  ```
Class java.util.concurrent.CountDownLatch was not found
Class java.util.concurrent.atomic.AtomicReference was not found
Class java.util.concurrent.locks.Lock was not found
Class java.util.concurrent.locks.ReentrantLock was not found
Class java.util.concurrent.locks.Lock was not found
Method java.lang.Class.getGenericInterfaces()[Ljava/lang/reflect/Type; was not found
Class org.slf4j.impl.StaticLoggerBinder was not found
Method java.lang.ClassLoader.getSystemResources(Ljava/lang/String;)Ljava/util/Enumeration; was not found
Class org.slf4j.impl.StaticLoggerBinder was not found
Class java.util.concurrent.atomic.AtomicReference was not found
Class java.util.concurrent.CountDownLatch was not found
Method java.io.PrintStream.println(F)V was not found
  ```
</details>

I just wanted to check if ODE4J can be compiled through TeaVM. Apparently not.

closed 2020-08-01 by zeganstyl

zeganstyl, 2020-08-07:
I checked, ode4j can be compiled to javascript through TeaVM.
You can check here (physics section is last):
https://zeganstyl.github.io/thelema-teavm-tests
There is a problem with the trimesh, but I will try to fix it later.
Trimesh works on desktop well. There is same tests for desktop:
https://github.com/zeganstyl/thelema-engine/releases/tag/0.4.0

Problems with java.util.concurrent, I think, must be fixed on TeaVM side.
In ode4j needed only remove slf4j from core.

tzaeschke, 2020-08-08:
Thanks for letting me know, this is quite impressive :-)
The problem really is that this that removing slf4j would be a breaking change, it would at least require a major version bump. 
I assume you want to build some kind of 3D javascript library, potentially even a game or simulator. I just wonder how often you really need to recompile ode4j -> javascript. Ode4j is not changing much recently, so it sounds like compiling it when necessary from a locally modified copy or even maintaining a fork may be feasible?

zeganstyl, 2020-08-09:
I think, I can fork or even include all ode4j core into the thelema-ode4j module.
As I understand, TeaVM can only produce single javascript file `classes.js`, like [this](https://github.com/zeganstyl/thelema-teavm-tests/blob/master/classes.js). Whole application compiles not so long.
And I think it is better to include ode4j core to module, so it can be easily distributed, modificated and tested.
I will provide a link to ode4j by BSD license.