tzaeschke/ode4j #73: Setting axis to Y-up

jfelrod1960, 2021-02-18:
Is there a way to set ode4j to use a y-up orientation?

tzaeschke, 2021-02-22:
Do you know whether this is possible in ODE C++? 

Also, I think odej4/ODE does not really care, unless you look at rotation matrices?

tzaeschke, 2021-02-23:
What exactly do you want to achieve?

I don't think ode/ode4j has a concept of 'up'.

For gravity, it is easy to set positive or negative gravity.
If you want some kind of general rotation, I suppose you can specify a rotation matrix for your ground floor (or some other fixed body) and use `dBodyVectorFromWorld` / `Body.vectorFromWorld` to invert the y-axis. But I guess it is faster and easier just to invert the y-axis manually.

Maybe you can ask on the C++ forum? ode4j supports almost everything that ODE C++ supports (except things OPCODE).

closed 2023-05-16 by tzaeschke