tzaeschke/ode4j #74: DPlane.getQuaternion() fails with NullPointerException

i-make-robots, 2021-12-24:
```
public class ODEPhysicsTests {
	@Test
	public void planeGetQuaternionFail() {
		OdeHelper.initODE2(0);
		DSpace space = OdeHelper.createHashSpace(null);
		DPlane plane = OdeHelper.createPlane(space,0,0,1,0);
		DQuaternionC q = plane.getQuaternion();
	}
}
```
if getQuaternion exists for all DGeom it should *succeed* for all DGeom or documented that it is known to fail deliberately.

tzaeschke, 2022-11-30:
Apologies for the delay.

This now throws an exception with a dedicated error message, as intended. There may already have been a message reported, but only to the active logger and with visibility "DEBUG".

tzaeschke, 2023-04-27:
This should be fixed now in master. Fix was added with #86, test was added with #78.

tzaeschke, 2023-04-27:
Thanks again for reporting!

closed 2023-04-27 by tzaeschke