tzaeschke/ode4j #76: DTriCallback is unused

Gaming32, 2022-11-27:
After setting up a `DTriCallback`, I found that it does nothing. Looking into the code, it's never called.

Gaming32, 2022-11-27:
Looking a bit more into this, is this covered by #5?

tzaeschke, 2022-11-28:
Yes, it would be covered by #5, except that this is unlikely to happen anytime soon (unless someone volunteers to do it).

Maybe you can use `DTriRayCallback` instead?

tzaeschke, 2022-11-28:
I do not know the reason for this, it maybe an oversight. I think it may be easy to implement. Do you have a good workaround.

While it should be easy to do, it would be another deviation from the original ODE, which I try to keep at a minimum. If you cannot find a good + easy workaround, let me know and we can add it.

Gaming32, 2022-11-28:
I did find a workaround. I separated my existing trimesh into separate trimeshes for each collision type, so I can check which trimesh was collided with to get the collision type, as opposed to looking up the collision type for the specific tri I collide with.

tzaeschke, 2022-11-30:
@Gaming32 I implemented a fix + tests, can you have a look whether it works for you?

tzaeschke, 2022-11-30:
> The biggest problem, in my opinion, is that `Callback` is a required argument for creating a `DTriMesh`, but then it's unused.

You should be able to use `OdeHelper.createTriMesh(space, data, null, null, null);`.

tzaeschke, 2023-05-25:
Trimesh callbacks have been implemented (except ArrayCallback, which has been deprecated).
OdeHelper now has a method to create trimeshes withou requiring callback arguments.
See PR #77

closed 2023-05-25 by tzaeschke