tzaeschke/ode4j #127: Weird behavior in DemoCrash

tzaeschke, 2023-09-09:
There are two issues, both point to a (two?) problems in the collider:

- Letting DemoCrash sit still for a few second will occasionally show the box stack becoming "active", as if one of the boxes moved slightly or was brushed by another object.
-  Choosing `ReorderingMethod.REORDERING_METHOD__DONT_REORDER` in `DxQuickStep` line 147 causes DemoCrash to become unstable. 

Both could be a good starting point for debugging a problem in DxQuickStep.

Murat65536, 2024-11-17:
The box stack is becoming active because it is being brushed by the centipede object. If you print out the position of the sphere that is brushing the box stack, you will see that it's position is slightly changing, which is probably causing the the box stack to become "active".

tzaeschke, 2024-11-17:
Nice, thanks for pointing that out.

tzaeschke, 2025-04-26:
With the port of 0.16.6 (and the fix in DxQuickStep) the problem is not reproducible anymore, except for very large walls (e.g. 120 x 50). The default RANDOM_REORDER has no problem at all.

closed 2025-04-26 by tzaeschke