tzaeschke/ode4j #136: Cannot download sources

i-make-robots, 2024-04-23:
When I look at a method in ODE4J I get (for example)

```java
public interface DCylinder extends DGeom {
    void setParams(double var1, double var3);
```

but in the original source I see

```java
public interface DCylinder extends DGeom {
	void setParams (double radius, double length);
```

which is way more informative.  If I had source then maybe it would have kept `radius` and `length?

Also where is it worth it for me to add method description comments so the API is more intuitive?

tzaeschke, 2024-04-23:
Hi, thanks for reporting.
Can you tell me where you see `void setParams(double var1, double var3);`? Is it from a decompiler in your IDE?

source-jars are available from Maven, e.g. https://repo1.maven.org/maven2/org/ode4j/core/0.5.2/
Would it solve the problem if you import these into your IDE?

Apologies if I misunderstand the problem.

i-make-robots, 2024-04-23:
Yes, exactly.  when i try "get sources" in IntelliJ IDEA it says they can't be found.

i-make-robots, 2024-04-23:
Oh yeah, switching back to 0.5.2 worked.  Sorry for the confusion...

closed 2024-04-23 by i-make-robots